jj1bdx/airspy-fmradion #1: Multi-channel decoding (new feature)

iroks, 2019-04-07:
is it possible to make the stereo FM-decoding for the multiple stations at the same time?

There is a software tool https://github.com/pvachon/tsl-sdr/wiki/Using-MultiFM that enables to configure multiple filtered output from a wideband stream. The typical configuration looks like:

  "device" : {
    "type" : "rtlsdr",
    "deviceIndex" : 0,
    "dBGainLNA" : 30
  },
  "sampleRateHz" : 2400000,
  "centerFreqHz" : 104600000,
  "nrSampBufs" : 128,
  "decimationFactor" : 12,

  "channels" : [
    {
      "outFifo" : "/tmp/ch1.out",
      "chanCenterFreq" : 104600000,
      "signalDebugFile": "/tmp/ch1.iq"
    }

  ]
}
...

Multifm supports only basic fm-demodulation but can output filtered streams to FIFO ("signalDebugFile"-option). I changed the filter from multifm and was able to decode and play the resulted stream (from signaldebugfile-fifo) using Matlab or GNU Radio (incl. Stereo-FM decoding and RDS). 

The idea is following: if you could extend airspy-fmradion with a support of a virtual fifo device (or just file input) with configurable sample rates that could be really great. If you could support the provided format from multifm (Qf-16 bit) than it will be ideal. In such case, single FM-decoders can be started as separate processes and the resulted streams can be sent for the processing to the further pipe; e.g. ffmpeg or ices. 

Thanks in advance for your consideration.

jj1bdx, 2019-04-07:
Hi @iroks: so I guess what you want me to do is to make stereo decoding section working separately without invoking the frontend SDR device. Maybe an interesting application. Currently, airspy-fmradion expects 300kHz to 384kHz for the demodulator, though it will even work in a higher-frequency configuration. I'll think about adding a feature for including ~400kHz IF, or even an MPX audio signal. Allow me some time.

iroks, 2019-04-07:
jj1bdx,
thanks for the fast reply. Yes, you are correct that the proposal to make stereo decoding section working separately without invoking the frontend device. There are following scenarios:
- Decoding from raw IQ file
- Decoding from stdin (piping support)
- Decoding from fifo socket and may be network sockets

It would be really great if airspy-fmradion could support different coding format, e.g. 8 bit, 12 bit, 16 bit (incl. Q15 as in the above mentioned multifm ) out of the box. Support of different sample rates could be very helpful as well to use the host resources more efficient. 

This feature will enable a lot of interesting scenarios, e.g. multi-channel FM decoding, support of new hardware (e.g. PlutoSDR) without frontend integration, network support for FM decoding etc.

Thanks in advance.

iroks, 2019-08-27:
any news or plans regarding the implementation?

iroks, 2019-08-27:
many thanks in advance

jj1bdx, 2019-08-27:
@iroks sorry being so late. The first step will be receiving data from stdin (preferably from `airspyhf_rx`, `airspy_rx`, and `rtl_sdr`.

atymic, 2020-03-15:
Hey all!

Just wondering if there was any progress on this?
I take it that's it's possible to tune multiple stations at once with the available bandwidth, just not supported yet?

Thanks for all your great work!

jj1bdx, 2020-12-09:
At this moment on 9-DEC-2020, I will drop this feature request.
While this feature looks interesting, making this feature a practical one will require a capability to receive data from an IQ stream server such as spyserver or other stream servers.
Also, airspy-fmradion should be kept as a modular tool. I like SDRsharp and SDR-Console, and also gqrx, but airspy-fmradion is not something like those pieces of GUI software.

closed 2020-12-09 by jj1bdx

iroks, 2020-12-09:
is it possible to add 32 bit support as well?

currently sox is used for decoding but it is not practical because ADC supports upto 768 kHz that is not supported by sox.
current fifo pipe looks like:

sox -r 96000 -e unsigned -b 32 -c 2 -t raw /tmp/ch5.raw -r 96000 -e unsigned -b 24 -c 2 -t raw - | airspy-fmradion -m AM -f medium -t FileSource -c filename=-,format=S24_LE,raw,srate=96000,lgain=0,mgain=0,vgain=0 -R - | play -t raw -esigned-integer -b16 -r 48000 -c 1 -

jj1bdx, 2020-12-09:
[FileSource.cpp has the type `FLOAT` defined](https://github.com/jj1bdx/airspy-fmradion/blob/master/sfmbase/FileSource.cpp#L11).
You might be interested in airspy-fmradion's `-F` option, which generates 32bit float output for the output audio.

jj1bdx, 2020-12-09:
Also, if you need to play audio direct from airspy-fmradion, try using `-P` option, to specify the PortAudio device.

iroks, 2020-12-09:
actually I interested in the input; I get from ADC 2x32 bit (32 bit I/32 bit Q); if understand correct, only 24 bits are supported at the moment.

int FileSource::to_sf_format(FormatType format_type) {
  int ret = 0;

  switch (format_type) {
  case FormatType::S8_LE:
    ret = SF_FORMAT_PCM_S8;
    break;
  case FormatType::S16_LE:
    ret = SF_FORMAT_PCM_16;
    break;
  case FormatType::S24_LE:
    ret = SF_FORMAT_PCM_24;
    break;
  case FormatType::U8_LE:
    ret = SF_FORMAT_PCM_U8;
    break;
  case FormatType::Float:
    ret = SF_FORMAT_FLOAT;
    break;
  default:
    assert(0);
    break;
  }

jj1bdx, 2020-12-09:
If the ADC outputs a linear 32-bit PCM stream, the format should be translated to something which airspy-fmradion accepts. I don't think I will add a 32-bit linear PCM anything soon, because It would require IEEE 64-bit double float internal processing. Internally, airspy-fmradion processes all IF stage in IEEE 32-bit float (before the demodulator). I have no intention to expand it to IEEE 64-bit double float, due to the possible computational complexity, and lack of library support.

iroks, 2020-12-09:
many thanks for the explanation