jj1bdx/airspy-fmradion #2: SoapySDR support (New Feature)

nmaster2042, 2019-05-18:
Do you plan to add SoapySDR support ?

It would bring access to more sdr devices, like funcube dongle, sdrplay, etc.

jj1bdx, 2019-05-19:
Hi @nmaster2042:
I've been told about this at https://github.com/jj1bdx/ngsoftfm-jj1bdx/issues/1 a long time ago.
So far I need to find out which format (float 32bit + signed 16-bit + unsigned 8-bit integer formats are needed so far) to support the streamed IQ data types. I've found SoapySDR library does not automatically convert between these data types.
I know the support for SoapySDR might be useful, so allow me some time.

jj1bdx, 2020-12-09:
At this moment on 9-DEC-2020, I will drop this feature request.

closed 2020-12-09 by jj1bdx