jj1bdx/airspy-fmradion #3: Cannot change sample rate

montgomeryb, 2019-06-05:
I found out from reviewing the code, that the '-r' option has been removed. 

I'm sure there's a good reason for this, but it took me by surprise when I tried to experiment to see if different sampling rates would change the quality of the output.

jj1bdx, 2019-06-05:
I've removed the -r option because fixing the sampling rate to 48kHz simplified the rate conversion ratio issues and eased the design constraints. After all, FM stereo output does not exceed the 15kHz bandwidth limit, let alone other AM/SSB output (4.5kHz max). Also, I'm developing this on macOS mostly, so using the sox utility is a requirement; if you need a sampling rate conversion of the output, use the sox functionality.

jj1bdx, 2019-06-05:
This is the commit I changed the `pcmrate` to the fixed value: https://github.com/jj1bdx/airspy-fmradion/commit/22121d08b739ee5482912f19fdfa9cdd17127655

jj1bdx, 2019-06-05:
And I've found that the usage help message of main.cpp still contained the invalid `-r` option. My bad.
Fixed the usage message in v0.6.9 at https://github.com/jj1bdx/airspy-fmradion/commit/498d34d7f0f2b1ed49c35ab9dd4c3682521d7a29. Thanks for reporting and sorry for the inconvenience.

closed 2019-06-05 by jj1bdx

montgomeryb, 2019-06-05:
> Fixed the usage message in v0.6.9 at 498d34d

No problem. Thanks for the quick response - and your work. I'm finding this
project very helpful.

Bryan.