jj1bdx/airspy-fmradion #9: Raspberry Pi evaluation

jj1bdx, 2019-09-08:
There would be no show-stopper for Raspberry Pi on airspy-fmradion, because no architecture-dependent libraries are used either on Linux or macOS. Operating airspy-fmradion on Ras Pi is technically feasible and should be pursued for the real-world application.

jj1bdx, 2019-09-08:
Teruhiko Hayashi, JA2SVZ, has kindly given me feedback on running airspy-fmradion on his Raspberry Pi 4B, and comparing his state-of-the-art FPGA FM Tuner FMDDC-3. Here's his preliminary report in PDF on how the multipath canceling filter works on airspy-fmradion, with Airspy HF+ Discovery:

[Airspy_fmradion_MPC.pdf](https://github.com/jj1bdx/airspy-fmradion/files/3587303/Airspy_fmradion_MPC.pdf)

jj1bdx, 2019-09-08:
[Airspy_fmradion_MPC.pdf](https://github.com/jj1bdx/airspy-fmradion/files/3587304/Airspy_fmradion_MPC.pdf)

jj1bdx, 2019-09-08:
[image]
[image]
[image]

montgomeryb, 2019-10-11:
I am running it on a pi 3B, along with openvpn. I pipe it to lame and to ezstream for streaming on icecast2. The average load on the pi with all of this is 0.75

Incidentally, I built most of the packages from source as the packaged versions were not recent enough.

jj1bdx, 2019-10-11:
@montgomeryb Many thanks for the load average report. I also think that some packages, especially the Airspy R2/HF+ drivers, might be too old.

jj1bdx, 2019-12-16:
Ras Pi 3B with NEON-enabled libvolk resulted in up to `-E 30` of the multipath filter capability with Airspy HF+ Dual Port (in 384kHz IF). Not the best, but being able to use the sample data up to 234µs (of the past data) echo is much much better than being unable to use any.

jj1bdx, 2020-01-03:
@jj1bdx I guess I can close this issue because I've done Ras Pi 3B evaluation myself and other people including @bstalk have already done an extensive evaluation of the performance on Ras Pi 4B.

closed 2020-01-03 by jj1bdx