jj1bdx/airspy-fmradion #10: VOLK (libvolk) for vector computing

jj1bdx, 2019-10-04:
Using [VOLK (libvolk)](http://libvolk.org/) makes the processing much faster, especially for the CPU-intensive functions such as the multipath filter. [The libvolk branch](https://github.com/jj1bdx/airspy-fmradion/tree/libvolk) is set for the migration.

Kudos to Takehiro Sekine for the suggestion!

iroks, 2019-10-04:
you can also consider the usage of GPU; please take a look at this paper https://www.ncbi.nlm.nih.gov/pmc/articles/PMC5695887/

jj1bdx, 2019-10-14:
@iroks I have no testing environment of GPU right now. I'm writing this software mostly by Mac mini 2018 with no external GPUs. BTW the paper looked informative. Thx!

jj1bdx, 2019-10-20:
Ref: https://github.com/gnuradio/volk/pull/291

jj1bdx, 2019-10-27:
VOLK seems stable both on macOS and Ubuntu, both on arm and x86_64. From v0.8.x libvolk is required.

closed 2019-10-27 by jj1bdx