jj1bdx/airspy-fmradion #11: Exit on Write Error??

montgomeryb, 2019-10-11:
I am running airspy-fmradion in a script, piping the output to lame and then to ezstream. This script is installed as a systemd service.

I have issues where at some point the process breaks and I get 'ERROR: AudioOutput: write failed (Broken pipe)'.

I see this is in main.cpp. Does it make sense to have the program either default to exiting, or provide an option for the program to exit in this case? I was planning to add stop_flag.store(true); to the condition in write_output_data - but I have yet to try this.

Maybe there's another way I can catch this error and exit the program?

jj1bdx, 2019-10-11:
@montgomeryb Thanks for the usage report, especially about invoking the program from systemd.

I haven't really checked out the detailed cause of the broken-pipe exit code. I'd be glad if you could try adding the `stop_flag.store(true)` and take a look at how the behavior changes.

montgomeryb, 2019-10-11:
I think the issue was a timing error, my systemd script that I created did check for an open port with netcat, but it seems that maybe that wasn't enough to determine if icecast is fully up and accepting streams.
I made the change to `main.cpp`:
```
    if (!(*output)) {
      fprintf(stderr, "ERROR: AudioOutput: %s\n", output->error().c_str());
      fprintf(stderr, "BRM: Setting stop_flag to true.\n");
      stop_flag.store(true);
    }
``` 
With that, it exits airspy-fmradion, the script then exits from systemd and systemd restarts it. By now, icecast2 is fully up and the process works as intended. 
I've since updated my script to actually use curl, and that seems a better indicator that icecast is up than just using netcat to check the port is open.

Thanks for your work 👍

jj1bdx, 2019-10-14:
@montgomeryb Committed as https://github.com/jj1bdx/airspy-fmradion/commit/7b322b87cc16a896e7a5ad9c22dd1332257e11a2 to the libvolk branch for the time being.

jj1bdx, 2019-10-26:
Merged to v0.8.0-pre0.

closed 2019-10-26 by jj1bdx

jj1bdx, 2021-11-01:
@montgomeryb I guess https://github.com/jj1bdx/airspy-fmradion/commit/b12c6307301ff8aee538d408ce1efdfb2d338b0e#diff-608d8de3fba954c50110b6d7386988f27295de845e9d7174e40095ba5efcf1bb might have fixed the bug.
Please try 20211101-0.

montgomeryb, 2021-11-14:
Seems to be running fine. Nice work.