jj1bdx/airspy-fmradion #19: Possible to change fixed sample rate of 48000?

trs79, 2019-12-11:
If I wanted to use a higher audio sample rate then 48000, would it be as simple as modifying the `sample_rate_pcm ` constant before compiling?

jj1bdx, 2019-12-12:
@trs79 While technically changing the `sample_rate_pcm` might suffice, I think you need to consider the following issues:

* FM output is already filtered by ~18kHz (none at 19kHz to remove the stereo pilot tone of 19kHz)
* AM output is ~6kHz at maximum (the sampling rate is 12kHz as in `AmDecoder::internal_rate_pcm`)
* If you just want to change the output rate, using `sox` for the output be far easier than modifying the source code
* The ratio of FM IF (384kHz) and the audio output should be an integer, otherwise the conversion process might take more CPU processing power

stowellt, 2019-12-12:
Thanks for the response. I was asking because I've been using a ADS Instant FM USB dongle with the si470x chipset, and it outputs 96KHz. I'd like to replace it with a RTL-SDR and was hoping to match the output rate. I was hoping to keep the sample rate higher at the source, rather then using Sox to upsample the 48kHz output rate.

I'm a novice when it comes to frequency processing, so please forgive my ignorance. When you say FM output is already filtered by ~18kHz, how would that affect changing the audio sample rate? Thanks.

jj1bdx, 2019-12-12:
@stowellt Hi Tim: the technical specification of FM stereo broadcast defines that each of the left and right channels contains a frequency range of 30Hz to 15kHz. Many of the modern digital FM transmitters use 32kHz for the sampling frequency because the 32kHz rate is sufficient to transmit 15kHz audio. A sampling rate of X is sufficient to carry the maximum sampled frequency of X/2 (divided by two). So using the 48kHz rate is already overkill for the FM, let alone 96kHz, technically speaking. While a higher sampling rate does no harm for transmitting audio, it won't give you anything additional either, especially if the transmitted audio range is limited.

I have chosen 48kHz for the audio sampling rate because it is practically easiest to handle in computers among different architectures and easy to design when IF should be an integer multiple of the audio sampling rate. Raising the sampling rate of a 48kHz audio to 96kHz gives no additional sound quality, though it won't lose any either.

jj1bdx, 2019-12-12:
See [US FCC 47 CFR Section 73.310](https://www.law.cornell.edu/cfr/text/47/73.310) for a brief list of technical definitions.

trs79, 2019-12-14:
Thanks for the information and explanation! I have a better understanding now. I'll close this issue.

closed 2019-12-14 by trs79