jj1bdx/airspy-fmradion #20: Problems with make

wa2zkd, 2019-12-31:
I get through all the cmake stuff OK I think....  I see this:, but my interpreatation is that libsoxr does get found and it is there when I look:

-- Checking for module 'libsoxr'
--   No package 'libsoxr' found
-- libsoxr: /usr/local/include, /usr/local/lib/libsoxr.so

cmake ends with: 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/jim/Apps/airspy-fmradion/build

But when I try to make it I get this:

[  4%] Building CXX object CMakeFiles/sfmbase.dir/sfmbase/AirspyHFSource.cpp.o
/home/jim/Apps/airspy-fmradion/sfmbase/AirspyHFSource.cpp: In member function ‘bool AirspyHFSource::configure(int, uint8_t, uint32_t)’:
/home/jim/Apps/airspy-fmradion/sfmbase/AirspyHFSource.cpp:205:16: error: ‘airspyhf_is_low_if’ was not declared in this scope
   int low_if = airspyhf_is_low_if(m_dev);
                ^~~~~~~~~~~~~~~~~~
/home/jim/Apps/airspy-fmradion/sfmbase/AirspyHFSource.cpp:205:16: note: suggested alternative: ‘airspyhf_stop’
   int low_if = airspyhf_is_low_if(m_dev);
                ^~~~~~~~~~~~~~~~~~
                airspyhf_stop
and it fails.

What am I doing wrong?

jj1bdx, 2019-12-31:
@wa2zkd Install [the latest libairspyhf from the master branch](https://github.com/airspy/airspyhf). The older ones do not have `airspyhf_is_low_if()` defined in the library.

wa2zkd, 2019-12-31:
Thanks, I now have it built on a x86 Linux desktop

closed 2019-12-31 by wa2zkd