jj1bdx/airspy-fmradion #21: sample rate error at runtime

wa2zkd, 2019-12-31:
Not to be a newbie pest....   but..   it build OK and now see this when using one of the examples

(base) jim@Desktop:/$ airspy-fmradion -m am -t airspyhf -q     -c freq=666000     -b 0.5 -F - |     play --buffer=1024 -t raw -e floating-point -b32 -r 48000 -c 1 -q -
airspy-fmradion v0.8.2
Software FM/AM radio for Airspy R2, Airspy HF+, and RTL-SDR
output buffer length: 0.5 [s]
writing raw 32-bit float little-endian audio samples to '-'
play WARN alsa: can't encode 0-bit Unknown or not applicable
using device 0: Serial dd5297aa2fbf3448
ERROR: configuration: Invalid sample rate in AirspyHFSource::configure initialization

jj1bdx, 2020-01-01:
@wa2zkd Upgrade the Airspy HF+ firmware to the latest version [R3.0.6-BB](https://airspy.com/downloads/airspy-hf-flash-20190820-bb.zip), and please retry. Old Airspy HF+ does not support the default sampling rate of 192kHz for AM (and 384kHz for FM).

wa2zkd, 2020-01-01:
Thanks again... that fixed it

Happy New Year

closed 2020-01-01 by wa2zkd

jj1bdx, 2020-01-01:
@wa2zkd You're welcome and thanks for reporting the issue. I've made a fix to the README.md as in: https://github.com/jj1bdx/airspy-fmradion/commit/d1e32abdc36bdef766b53477ba4cb0aade54c5d9#diff-04c6e90faac2675aa89e2176d2eec7d8

73 es HNY de JJ1BDX