jj1bdx/airspy-fmradion #22: RPi 3 install troubles

wa2zkd, 2020-01-01:
I have successfully intsalled this on my x86 box, but never saw this issue with mako.  I think mako installed

pip install mako
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting mako
  Downloading https://files.pythonhosted.org/packages/b0/3c/8dcd6883d009f7cae0f3157fb53e9afb05a0d3d33b3db1268ec2e6f4a56b/Mako-1.1.0.tar.gz (463kB)
    100% |████████████████████████████████| 471kB 483kB/s 
Collecting MarkupSafe>=0.9.2 (from mako)
  Downloading https://files.pythonhosted.org/packages/b9/2e/64db92e53b86efccfaea71321f597fa2e1b2bd3853d8ce658568f7a13094/MarkupSafe-1.1.1.tar.gz
Building wheels for collected packages: mako, MarkupSafe
  Running setup.py bdist_wheel for mako ... done
  Stored in directory: /root/.cache/pip/wheels/98/32/7b/a291926643fc1d1e02593e0d9e247c5a866a366b8343b7aa27
  Running setup.py bdist_wheel for MarkupSafe ... done
  Stored in directory: /root/.cache/pip/wheels/f2/aa/04/0edf07a1b8a5f5f1aed7580fffb69ce8972edc16a505916a77
Successfully built mako MarkupSafe
Installing collected packages: MarkupSafe, mako
Successfully installed MarkupSafe-1.1.1 mako-1.1.0

**but when I tried to install libvolk, I get**

-- Python checking for mako >= 0.4.2
-- Python checking for mako >= 0.4.2 - not found
-- 
-- Python checking for six - python 2 and 3 compatibility library
-- Python checking for six - python 2 and 3 compatibility library - found
CMake Error at CMakeLists.txt:93 (message):
  Mako templates required to build VOLK

-- Configuring incomplete, errors occurred!

jj1bdx, 2020-01-02:
@wa2zkd Try `pip3 install mako`. The latest libvolk prefers Python 3 than Python 2 (already deprecated).

wa2zkd, 2020-01-02:
That solved that!

closed 2020-01-02 by wa2zkd

jj1bdx, 2020-01-02:
@wa2zkd Updated the document. https://github.com/jj1bdx/airspy-fmradion/commit/10be63a0bd1b35f3537bd0689815fefe4d7d4b7d
(Python 2 has already been phased out, so I guess more and more distribution will set the Python 3 as the default major Python distribution.)