jj1bdx/airspy-fmradion #23: Raspberry Pi 3 Model B Rev 1.2, runs but get alsa under-runs

wa2zkd, 2020-01-02:


wa2zkd, 2020-01-02:
May be related to an under-voltage issue on Pi. The Airspy draws a slug of current, dmesg shows an UV warning and the Pi's red LED goes out. I have tried various PSU and cables, same deal. I'll have to get a powered USB hub going next.

wa2zkd, 2020-01-02:
To test my theory, I connected a variable lab supply to the Pi and with the AS on a USB port, I adjusted the voltage on the USB port to 5.0V.  No more under-runs!  Note: ASHF+ draws about 325 mA

closed 2020-01-02 by wa2zkd

jj1bdx, 2020-01-03:
@wa2zkd I was using a 5V 2A dedicated power supply when I successfully ran airspy-fmradion on a Raspberry Pi 3B. The power supply voltage indeed affects the processing performance.