jj1bdx/airspy-fmradion #24: stereo decoder ?

jpuigs, 2020-05-20:
Hi,
I would like to use part of this project, just to make a stereo decoder, but I don't know if it's possible.
I want RDS and Stereo,  so I need to use rtl_fm command line and redsea decoder.
$ rtl_fm ......... | redsea -e

RTL_FM gives to redsea a MPX signal (171 or 192 KHz) which redsea, decodes RDS and gives output in json format.
Redsea -e parameter echoes the input (stdin) signal to stdout and print decoded groups to stderr, so what I would like is a simple stereo decoder which take MPX signal from redsea ouput (stdout),   decodes stsreo sound and give ouput in S16LE 2 channel output....

So command line would be like this...

**rtl_fm** -M fm -l 0 -A std -s 171k -F 9 -f 87.9M | **redsea** -e | **stereodecoder** | **aplay** -r 171k -f S16_LE -c 2

Is it possible to use some parts to get this stereodecoder?

jj1bdx, 2020-08-30:
@jpuigs The current (v0.8.5) FM Stereo decoder part of airspy-fmradion assumes 384ksamples/sec (though the rate might differ) 32bit-float-complex IQ signal stream. 192ksamples/sec is insufficient for the high fidelity playback. You might be interested in sfmbase/FmDecode.cpp and include/FmDecode.h for the details.

jj1bdx, 2020-08-30:
@jpuigs OTOH, I need to redesign the whole programs for splitting the FM decoder part as an individual software.

jpuigs, 2020-09-01:
ok, understood. 👍

jj1bdx, 2020-12-09:
At this moment on 9-DEC-2020, I will drop this feature request.

closed 2020-12-09 by jj1bdx