jj1bdx/airspy-fmradion #26: Slow audio

Akito01, 2020-06-15:
Using the following syntax causes slowed-down audio:

airspy-fmradion -t airspy -c freq=88000000,srate=6000000,mgain=12 -b 1.0 -R - | sox -t raw -r 40000 -es -b16 -c1 -V1 - test.wav rate 11025

I can adjust the audio speed bij adding the tempo or speed parameter in sox, but when using rtl_fm I don't need to use that parameter. How can I pipe the output to sox with the original speed?

jj1bdx, 2020-06-15:
@Akito01 The audio rate is fixed to 48000Hz. Use `sox -t raw -r 48000 ...` to process the output by sox.

Akito01, 2020-06-15:
@jj1bdx like that the audio is still slowed down. Including the pitch.
It's getting close to reality when setting -r to 80000 actually. So, how to correct this?

Also; it seems I can only get static noise above 108MHz. Ie. 126.2MHz is airport information here. With other tools I can listen to it using the airspy dongle. Using airspy-fmradion it's just static noise.

jj1bdx, 2020-06-15:
@Akito01 This is an example to listen to RJTT (Tokyo Haneda) ATIS here:
```sh
airspy-fmradion -f medium -m am -t airspyhf -c freq=128800000,srate=192000 -b 0.5 -F - | play --buffer=1024 -t raw -e floating-point -b32 -r 48000 -c 1 -q - compand 0.0001,0.2 26:-70,-60,-20,-20
```

jj1bdx, 2020-06-15:
@Akito01 And this is to listen to NHK-FM Tokyo 82.5MHz here. Note well that the output of FM broadcast reception is *stereo* (i.e., 2 channels), so use `-c 2` in sox.

```sh
airspy-fmradion -E100 -t airspyhf -q -c freq=82500000 -F - | \
  play -t raw -e floating-point -b32 -r 48000 -c 2 -q -
```

Akito01, 2020-06-17:
@jj1bdx great thank you, this has actually helped a lot.

By the way, I'm using it on a 2GB Raspberry Pi 4. After about 3 minutes, the airspy-fmradion process gets killed. Dmesg shows the following:
Out of memory: Kill process 25082 (airspy-fmradion) score 698 or sacrifice child
[77737.017564] Killed process 25082 (airspy-fmradion) total-vm:1509012kB, anon-rss:1439748kB, file-rss:4468kB, shmem-rss:0kB
[77737.271990] oom_reaper: reaped process 25082 (airspy-fmradion), now anon-rss:0kB, file-rss:0kB, shmem-rss:0kB

After about 1 minute of running, it has consumed about 30% of the entire RAM already.
How to make this process consume less RAM?

jj1bdx, 2020-06-17:
@Akito01 There are some Ras Pi 3 users (including myself) and 4 users running airspy-fmradion, but no report of OOM so far. One thing which should be noted is that they all run with Airspy HF+, of a slower sampling rate. I have no idea of why so much memory space is consumed.

About the 44kHz-width signal: airspy-fmradion is not a generic receiver. You can modify the code for NBFM for receiving a 44kHz-width FM signal.

jj1bdx, 2020-07-29:
No further response since 17-JUN-2020. Issue closed.

closed 2020-07-29 by jj1bdx

jj1bdx, 2020-08-29:
On OOM: use 2.5Mbps sampling rate for Airspy R2. 3Mbps will suffice for Airspy Mini. I've finally tested airspy-fmradion on Ras Pi 4 B with 2GB memory and confirmed excessive memory usage expansion when the sampling rate was set to 10Mbps. The following settings worked OK with CPU usage ~100% and no excessive memory usage expansion:
```
airspy-fmradion -t airspy -E20 -c freq=82500000,srate=2500000,lgain=8,mgain=5,vgain=10 -F - | nc <receiver> <port>
```

jj1bdx, 2020-10-09:
[Version 20201009-0](https://github.com/jj1bdx/airspy-fmradion/tree/20201009-0) has `-f wide` option for the `-m nbfm` mode, specifically designed for NOAA weather satellites (BW: 40kHz, deviation: +-17kHz). Worked fine with [noaa-apt](https://noaa-apt.mbernardi.com.ar) decoder and Airspy HF+ Dual Port.