jj1bdx/airspy-fmradion #27: Is this compatible with Raspberry Pi 3?

hjf, 2020-07-31:
I managed to build this but have a major problem. The CPU load for the process is around 135%, and after a few seconds it displays a warning that the Input buffer is filling up (system too slow).
Then the process dies silently. 
I watched "top" while this ran, and found that the RAM usage goes up pretty quickly. It must be running out of memory.

This is how I'm running it:
`
airspy-fmradion -t airspy -c antbias=on -c freq=137500000,srate=3000000,lgain=2,mgain=0,vgain=10 -M -b 1.0 -R -  | /usr/bin/sox -t raw -e signed -c 1 -b 16 -r 48000 - test.wav rate 11025`

I want to use this with WXtoIMG to receive APT weather satellites. I have it working with rtl_fm but I wanted to try if there was a difference with the Airspy

jj1bdx, 2020-07-31:
@hjf Hi - I've actually run airspy-fmradion on an Raspberry Pi 3B+ *with Airspy HF+* which has *lower sampling rate* and *lower CPU load in general*. Never really tested the software with Airspy Mini and Raspberry Pi 3B+ yet though.

jj1bdx, 2020-07-31:
I will close this issue now, and will add the disclaimer on using airspy-fmradion on faster sampling rate devices including Airspy R2 and Mini.

closed 2020-07-31 by jj1bdx