jj1bdx/airspy-fmradion #34: Error : this file format does not support pipe write.

Haz3lnut, 2022-02-02:
Per the Usage specs:

```
-W filename    Write audio data to RF64/WAV S16_LE file
                 use filename '-' to write to stdout
```

However, I try to pipe to ffmpeg with the following:
`airspy-fmradion -t airspyhf -q  -c freq=105300000,srate=384000 -b 1.0 -W - | ffmpeg -f s16le -ar 48000 -ac 2 -i - -c:a libfdk_aac -b:a 192k -ar 48000 -af "volume=5dB" -y out.aac`

It returns the following error:
```
airspy-fmradion 20211209-0
Software FM/AM radio for Airspy R2, Airspy HF+, and RTL-SDR
output buffer length: 1 [s]
writing RF64/WAV int16 audio samples to '-'
ERROR: AudioOutput: can not open '-' (Error : this file format does not support pipe write.)
```
Am I missing something?

jj1bdx, 2022-02-03:
Please use the `-R` option for RAW binary audio output to the pipe.
The WAV file output driver rewrites the file header periodically, so it does not work with a pipe.

Haz3lnut, 2022-02-03:
Thank you for such a quick reply. Yes, `-R` option worked just fine for the pipe. 

Suggest remove the stdout pipe option from -W to prevent further confusion.

jj1bdx, 2022-02-03:
Added the explanations to the help message and README explicitly stating pipe is not supported by the `-W` and `-G` options, made the new release 20220203-0, including the documentation fixes. Issue closed.

closed 2022-02-03 by jj1bdx