jj1bdx/airspy-fmradion #41: `volk_32fc_x2_s32fc_multiply_conjugate_add_32fc' is deprecated on VOLK 3.1.0

jj1bdx, 2023-12-12:
Maybe we need to fix this ASAP. 

```
[ 87%] Building CXX object CMakeFiles/sfmbase.dir/sfmbase/RtlSdrSource.cpp.o
/home/kenji/src/airspy-fmradion/sfmbase/MultipathFilter.cpp: In member function 'void MultipathFilter::update_coeff(IQSample)':
/home/kenji/src/airspy-fmradion/sfmbase/MultipathFilter.cpp:136:3: warning: 'volk_32fc_x2_s32fc_multiply_conjugate_add_32fc' is deprecated [-Wdeprecated-declarations]
  136 |   volk_32fc_x2_s32fc_multiply_conjugate_add_32fc(
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /usr/local/include/volk/volk_alloc.hh:18,
                 from /home/kenji/src/airspy-fmradion/include/SoftFM.h:30,
                 from /home/kenji/src/airspy-fmradion/include/MultipathFilter.h:23,
                 from /home/kenji/src/airspy-fmradion/sfmbase/MultipathFilter.cpp:31:
/usr/local/include/volk/volk.h:1895:61: note: declared here
 1895 | extern VOLK_API p_32fc_x2_s32fc_multiply_conjugate_add_32fc volk_32fc_x2_s32fc_multiply_conjugate_add_32fc __attribute__((deprecated));
      |                                                             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

jj1bdx, 2023-12-12:
Suggested fix: https://github.com/gnuradio/volk/pull/695

argilo, 2023-12-12:
By the way, the deprecated kernels will probably remain available for a long time, but you should use the new ones when they are available so that your application will avoid Undefined Behaviour that causes crashes on some architectures.

jj1bdx, 2023-12-12:
@argilo 
I'm testing the new API with add2 of v3.1.0 and I've found it worked OK with macOS but not on Ubuntu so I'm investigating it now. I will write more when I can tell the details.

jj1bdx, 2023-12-12:
(This is a copy of https://github.com/gnuradio/volk/pull/695#issuecomment-1851805075 )

@argilo After a few hours of trial, I have to conclude that:

volk_32fc_x2_s32fc_multiply_conjugate_add_32fc of VOLK v3.1.0 worked OK on macOS 14.2 (Homebrew) with airspy-fmradion MultipathFilter::update_coeff().
Either volk_32fc_x2_s32fc_multiply_conjugate_add_32fc or volk_32fc_x2_s32fc_multiply_conjugate_add2_32fc of VOLK v3.1.0 did not work (no compilation error, but the calculation failed), in generic kernel (no CPU optimization) on airspy-fmradion MultipathFilter::update_coeff() running on Ubuntu 22.04.3 x86_64 compiled with gcc version 11.4.0. I had to fall back to VOLK 3.0.0, which worked OK. Also, airspy-fmradion compiled with VOLK 2.5.1 of Ubuntu 22.04.3's apt repository worked OK.

jj1bdx, 2023-12-12:
Added docs to use VOLK 3.0.0 or 2.5.2 for Ubuntu 22.04.3 for airspy-fmradion 20231212-1.
I still don't know why the code works OK on macOS and fails on Ubuntu.

jj1bdx, 2023-12-12:
The bugfix issue will be followed-up on #42.

jj1bdx, 2023-12-13:
#42 is fixed by https://github.com/jj1bdx/airspy-fmradion/pull/43.
Conditional compilation code is added as https://github.com/jj1bdx/airspy-fmradion/commit/e52c30f5f336c608b1034d80594628c9caab1c26 in the dev branch.
Case closed.

closed 2023-12-13 by jj1bdx