jj1bdx/airspy-fmradion #59: Stops after a few seconds.

EricFROL, 2024-01-03:
Running on dietpi.
Starts fine, and without printing any error stops the output of audio, as fldigi stops receiving data.

Log:
```
airspy-fmradion -t airspyhf -q -m usb -c freq=13881000,hf_att=0 -P -
airspy-fmradion 20231216-0
Software FM/AM radio for Airspy R2, Airspy HF+, and RTL-SDR
VOLK_VERSION = 020502
ALSA lib pcm.c:2666:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.front
ALSA lib pcm.c:2666:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2666:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2666:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib pcm.c:2666:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib pcm.c:2666:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib pcm.c:2666:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround40
ALSA lib pcm.c:2666:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround41
ALSA lib pcm.c:2666:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround50
ALSA lib pcm.c:2666:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround51
ALSA lib pcm.c:2666:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround71
ALSA lib pcm.c:2666:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2666:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2666:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib confmisc.c:1369:(snd_func_refer) Unable to find definition 'cards.0.pcm.hdmi.0:CARD=0,AES0=4,AES1=130,AES2=0,AES3=2'
ALSA lib conf.c:5180:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5703:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2666:(snd_pcm_open_noupdate) Unknown PCM hdmi
ALSA lib confmisc.c:1369:(snd_func_refer) Unable to find definition 'cards.0.pcm.hdmi.0:CARD=0,AES0=4,AES1=130,AES2=0,AES3=2'
ALSA lib conf.c:5180:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5703:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2666:(snd_pcm_open_noupdate) Unknown PCM hdmi
ALSA lib pcm.c:2666:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2666:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2666:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm.c:2666:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
ALSA lib pcm_oss.c:397:(_snd_pcm_oss_open) Cannot open device /dev/dsp
ALSA lib pcm_oss.c:397:(_snd_pcm_oss_open) Cannot open device /dev/dsp
ALSA lib pcm_a52.c:1001:(_snd_pcm_a52_open) a52 is only for playback
ALSA lib conf.c:5670:(snd_config_expand) Unknown parameters {AES0 0x6 AES1 0x82 AES2 0x0 AES3 0x2 CARD 0}
ALSA lib pcm.c:2666:(snd_pcm_open_noupdate) Unknown PCM iec958:{AES0 0x6 AES1 0x82 AES2 0x0 AES3 0x2 CARD 0}
ALSA lib confmisc.c:160:(snd_config_get_card) Invalid field card
ALSA lib pcm_usb_stream.c:482:(_snd_pcm_usb_stream_open) Invalid card 'card'
ALSA lib confmisc.c:160:(snd_config_get_card) Invalid field card
ALSA lib pcm_usb_stream.c:482:(_snd_pcm_usb_stream_open) Invalid card 'card'
ALSA lib pcm_dmix.c:999:(snd_pcm_dmix_open) unable to open slave
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
playing audio to PortAudio default device: name 'default'
using device 0: Serial 2f528d8095444ad5
tuned for 13.881 [MHz]
Decoding modulation type: usb
IF sample rate: 384000 [Hz], IF decimation: / 8
Demodulator rate: 48000 [Hz], audio decimation: / 1
AM demodulator deemphasis: 100 [µs]
Filter type: default
```

jj1bdx, 2024-01-04:
You must use a loopback device (snd_aloop or snd-aloop) between airspy-fmradion and fldigi.
The whole setup must be on ALSA level because airspy-fmradion does not use PulseAudio at all.
You can find the PortAudio device number by [system_info.py of PyAudio](https://github.com/CristiFati/pyaudio/blob/master/examples/system_info.py).

I guess this is not an airspy-fmradion problem because your audio output is directed to PortAudio `default` device.

EricFROL, 2024-01-04:
Any advice on how to make airspy-fmradion output to also instead of pulseaudio?

EricFROL, 2024-01-04:
[image]
This is what I can select in fldigi

jj1bdx, 2024-01-05:
Use hw:2,0 for airspy-fmradion output (`-P 2` in your setting), and use hw:2,1 for fldigi input.
airspy-fmradion does not use PulseAudio at all.

jj1bdx, 2024-01-05:
fldigi
* open dialog box: Configure -> Config Dialog -> Soundcard -> Settings
* Set Sample rate to 48000Hz
Also:
* use hw:2,0 for airspy-fmradion `-P 1` (that's my bad)

EricFROL, 2024-01-07:
Now starts audio, but same issue as before, suddenly stops. Btw airspyfmradion command continues active

jj1bdx, 2024-01-08:
In my case, running Ubuntu 22.04.3 x86_64, airspy-fmradion 20240107-0, and fldigi 4.2.03.10 worked OK (at least the audio didn't stop). I don't think this is airspy-fmradion's general issue but rather a configuration issue.
Can you tell me your fldigi's version?

jj1bdx, 2024-01-08:
I spent six hours trying to find out that this is a bug in the PortAudio of Raspberry Pi. I even tried to rebuild my own PortAudio and traced the issue, but I decided I wouldn't be able to go further. Here in my Raspberry Pi environment, the snd_aloop device is recognized by the alsa programs (such as "aplay -L"), but the snd_aloop devices are not recognized at all by PortAudio. Other than this, airspy-fmradion and fldigi (on Raspberry Pi) both worked fine.

Possible related issues:

* https://github.com/PortAudio/portaudio/issues/431
* https://github.com/PortAudio/portaudio/issues/860

jj1bdx, 2024-01-08:
I installed a clean-slate Raspberry Pi OS 64bit (bookworm) to a Raspberry Pi 4B 2GB RAM, and necessary libraries with no private build other than airspy-fmradion, and found that everything went fine between the installed airspy-fmradion and fldigi. So I have to conclude there is something wrong with PortAudio on bullseye.

I conclude this is not an airspy-fmradion issue, so I will close this issue.

closed 2024-01-08 by jj1bdx

jj1bdx, 2024-01-08:
Further discussion: #69

EricFROL, 2024-01-08:
So what should I do?

EricFROL, 2024-01-08:
Recompiled pulseaudio and looks like now its working... weird tho... 
Thank you so much for your support, appreciate that.

jj1bdx, 2024-01-09:
@EricFROL I'm sorry, I can't propose any viable solution right now because it's out of the scope of airspy-fmradion.
The Debian bullseye problem of being unable to handle snd_aloop of PortAudio prevented me from debugging this case. 
As you told me privately, I hope your problem has been solved. I would consider terminating support for Raspberry Pi bullseye, due to older CMake and the whole issue I experienced yesterday (which I could not solve even by reinstalling PortAudio).