jj1bdx/airspy-fmradion #78: Build issue on Arch

zygmund2000, 2024-03-30:
Hi

I have Arch Linux OS and I'm trying to build latest version but some errors occurs:

```
-- volk 3.1: /usr/lib/libvolk.so
-- libairspy 1.0: /usr/include/libairspy, AIRSPY_LIBRARY-NOTFOUND
-- libairspyhf 1.7: /usr/include/libairspyhf, AIRSPYHF_LIBRARY-NOTFOUND
-- librtlsdr 2.0.1: /usr/include, RTLSDR_LIBRARY-NOTFOUND
-- libusb 1.0.27: /usr/include/libusb-1.0, /usr/lib/libusb-1.0.so
-- sndfile 1.2.2: /usr/lib/libsndfile.so, /usr/include
-- sndfile MP3 support enabled
-- libportaudio 19: /usr/include, /usr/lib/libportaudio.so
-- PortAudio libs: /usr/lib/libportaudio.so;/usr/lib/libasound.so;/usr/lib/libm.so;/usr/lib/libpthread.a
-- Configuring done (0.3s)
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
AIRSPYHF_LIBRARY
    linked by target "sfmbase" in directory /home/roygbiv/test/airspy-fmradion
AIRSPY_LIBRARY
    linked by target "sfmbase" in directory /home/roygbiv/test/airspy-fmradion
RTLSDR_LIBRARY
    linked by target "sfmbase" in directory /home/roygbiv/test/airspy-fmradion

```

I have all those libraries but still not found 

```
ls /usr/include/libairspy
.rw-r--r-- root root 9.7 KB Fri May 28 17:25:32 2021  airspy.h
.rw-r--r-- root root 4.7 KB Fri May 28 17:25:32 2021  airspy_commands.h
.rw-r--r-- root root 2.7 KB Fri May 28 17:25:32 2021  filters.h
.rw-r--r-- root root 1.7 KB Fri May 28 17:25:32 2021  iqconverter_float.h
.rw-r--r-- root root 1.6 KB Fri May 28 17:25:32 2021  iqconverter_int16.h
```

jj1bdx, 2024-03-31:
You need to install static libraries for AirSpy HF+, Airspy, and RTL-SDR, namely libairspyhf.a, libairspy.a, and librtlsdr.a.

jj1bdx, 2024-03-31:
I'm testing using shared libraries for Airspy HF+, Airspy, and RTL-SDR.
See #79

jj1bdx, 2024-03-31:
I will eventually merge #79 fix into main, but so far you can use this code: https://github.com/jj1bdx/airspy-fmradion/tree/5721407b3e98e8723a28898c75db8a9c1d604725

jj1bdx, 2024-03-31:
And please use libairspyhf *1.8* or later.

jj1bdx, 2024-03-31:
I assume that Arch Linux does not install static libraries for airspy, airspyhf, and rtl-sdr.

zygmund2000, 2024-03-31:
In past I was able to build fmradion with this os, so why can't you change it to use dynamic like other software?

zygmund2000, 2024-03-31:
> I will eventually merge #79 fix into main, but so far you can use this code: https://github.com/jj1bdx/airspy-fmradion/tree/5721407b3e98e8723a28898c75db8a9c1d604725

This commit buids without errors

jj1bdx, 2024-03-31:
@zygmund2000

> In past I was able to build fmradion with this os, so why can't you change it to use dynamic like other software? Rebuild Arch packages just for static is not a solution because every time when upstream updates comes it must be rebuild again.

I don't use Arch Linux by myself, and I have no resources to test all the different Linux distributions other than those listed in CHANGES.md.

jj1bdx, 2024-03-31:
The modification commit is already in the dev branch anyway, and will eventually be merged into the main branch (presumably at the next release). I will close this issue.

closed 2024-03-31 by jj1bdx